avikde/koduino #2: F4 support ?

rogerclarkmelbourne, 2015-01-07:
I notice in one of your commits 64f6560a25aa5a0dc259546ab227fc14eb5c87d1  that you mention that F4 works but not the libraries

Can you commit the boards.txt that has the F4 board in it?
As I'd like to test on my Discover 407VGT

avikde, 2015-01-07:
Hi Roger,

Sorry, but I'm still away and haven't tested any F4 chips yet. I just
checked that my code now *compiles* (that was the commit you saw) -- and it
now does with SPI and Wire as well. I will be sure to comment on this issue
after trying a few tests of my own on
* F405RG custom board, and
* F411RE nucleo board
and at that point it would be great to get some help testing on the F4
discovery.

Thanks,
-Avik

rogerclarkmelbourne, 2015-01-09:
Hi Avik.

No worries

BTW. Where did you get your source files for the standard peripheral libraries ?

It looks like the same files as in em:blocks ?

I'm not sure if this is the same as STM's own CPAL, but I've stayed away from CPAL because STM's license on the code was vague and it looked like you could not republish or distribute their code, you could only use it in your own commercial or private projects.

Also, I had an email from someone else looking for a Arduino STM32 system, but who didn't want to use libmaple, so I have sent him a link to your repo ;-)

PPS. I may fork your repo and see if I can add the F103 files, e.g if you are using the ones from em:blocks, I can use the copy of em:blocks that I have for these files

Cheers

Roger

avikde, 2015-01-13:
Hi Roger,

I just wanted to let you know that hasn't fallen off radar...I'm swamped
with work for the next couple of weeks but here are a couple of quick
answers:

I got the ST std peripheral libraries from ST's website. They are slightly
hard to find because that website is terrible, but here is the one for F4
<http://www.st.com/web/en/catalog/tools/PF257901>, for instance. I haven't
used (or at least, I did at some point but stopped using) the CPAL
libraries (which are a layer on top, as far as I understand). Good point
about the license, but I think it's fine based on
a) my (non-lawyer) reading of the comment on top of the source files--seems
MIT-ish, and
b) the spark core firmware
<https://github.com/spark/firmware/blob/master/src/spark_wiring_i2c.cpp> uses
it as well, and I trust that they wouldn't screw up something like that.

The F1 libraries from ST are in the repo already
<https://github.com/avikde/koduino/tree/master/stm32/system/STM32F10x_StdPeriph_Driver>,
but I haven't checked yet that my arduino -> ST interface code
<https://github.com/avikde/koduino/tree/master/stm32/cores/arduino> compiles
with the F1 libraries. I will write up a guide on making new variants very
soon, but I'm a little hard pressed for the next week or two, and wanted to
get this email out now.

I'll keep this issue open and comment once I have the "making new variants"
guide up, or I've had a chance to test the compilation myself.

Sorry about the delay!
-Avik

rogerclarkmelbourne, 2015-01-13:
Thanks Avik

I'm busy at the moment as well. I work freelance, and my clients are now coming back from their holidays, so my workload is increasing.

I suspect I won't have much free time until the end of January.

PS. I'd noticed the F103 files were in the Repo. I was going to add them to my fork and send you a pull request if not ;-)

Doing a "making new variants" guide is a really good idea. Then you won't need to do it all yourself.

I find with my repo that not many other people actually submit bug fixes, but plenty of people are happy to tell me whats wrong ;-)

avikde, 2015-11-02:
Now the F405 is pretty fully supported. It will take a small amount of work to port to other chips though, which I'll get to slowly / as there are boards that want this library.

closed 2015-11-02 by avikde