avikde/koduino #3: Nucleo STM32F334

rogerclarkmelbourne, 2015-03-19:
Hi Avik

I now have a Nucleo STM32F334, so you think I should be able to get your code to run on this board ?

Or are their significant differences between the F334 and the  F373

I've not looked at the spec, and they seem quite similar but the 334 has less ram (I can't see that would be an issue)

Is it worth giving it a go, or can you think of a reason why its not likely to work if I select the F373 board (as I can worry about ram overflow myself ;-)

avikde, 2015-03-19:
Hi Roger,

Hate to be the bearer of bad news but it looks like it is quite different
from the F373. :( The F3 series is especially fragmented: ST already
provides a completely different standard peripheral library for the F37x vs
F30x (as you can see here
<https://github.com/avikde/koduino/tree/master/stm32/system>) and it seems
like the F334 will need another one as well. Luckily the F4 and F1 series'
at least have the same standard peripheral library, and I've begun mucking
around with the F1 stuff. I'm sorry but I think that chip will take quite a
bit of work. :(

I got a Nucleo F411 myself BTW and that at least blinks an LED, and I've
been chipping away at other F4xx libraries (just got Wire working well, SPI
is up next). If you have any F4 boards lying around that may be the
quickest to get up and running (as I was hinting at Andrew Kroll in the
other email). I'm starting to assemble porting instructions here
http://avikde.me/koduino/html/md__porting.html

I also got that ebay maple mini clone you mentioned and while that's lower
priority for me I'll get to it eventually as well. :)

-Avik

rogerclarkmelbourne, 2015-03-20:
Hi Avik

No worries. 

I have some code for the F303 for my repo  (that someone provided), but its based on a much older version of libmaple, so I've had to put it in a completely separate section of my repo.

I'll need to compare the F303 with the F334 to to see if the peripheral access looks similar.

PS. Thanks for the porting instructions. I'm sure a F103 version would get a lot of interest, as 95% of the people on the Arduino forum seem to have those boards (because they are cheap and widely available)

avikde, 2015-11-02:
F301 now works quite well (and the code / pinout is even quite interchangeable with F302). Not gotten to F334 yet though.