avikde/koduino #4: You repo will not work with Arduino 1.6.2 - nor does mine :-(

rogerclarkmelbourne, 2015-03-31:
Hi Avik

You probably have not noticed, but Arduino 1.6.2 made a major change, and the ARM compiler is not installed by default, you need to go to the new Boards Manager under Tools->Boards and install the Due to get it to install the ARM compiler

But even when you do that, the repo won't work because they have moved where the compiler located :-(

You could change you platforms.txt to match, the new location, but you will need to add a read me to give instructions about installing the Due

But basically the board manager is the new method to install boards, but currently it only supports downloading the boards package data from here

http://downloads.arduino.cc/packages/package_index.json

See my issue about adding other third party boards via the Boards manager  https://github.com/arduino/Arduino/issues/2856

avikde, 2015-11-02:
It seems like there was some info in this forum post. 
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/39#issuecomment-88439318

I want to get to this, but I'm concerned that "packaging up" the code into a downloadable format for the json file makes it less friendly to version control.

rogerclarkmelbourne, 2015-11-02:
Avik,

Things have moved in a bit since i posted that issue, and Im going to remove support for versions earlier than 1.6.3 fairly soon

Re JSON

Yes. I agree, its not git friendly.

The solution appears to be, to use the github releases mechanism, which allow scripting to create downloadable files.

However I have not had time to do this, and some users are very opposed to this method of deployment as it hides the core files, which they may want to modify

rogerclarkmelbourne, 2015-11-02:
Avik,

Things have moved in a bit since i posted that issue, and Im going to remove support for versions earlier than 1.6.3 fairly soon

Re JSON

Yes. I agree, its not git friendly.

The solution appears to be, to use the github releases mechanism, which allow scripting to create downloadable files.

However I have not had time to do this, and some users are very opposed to this method of deployment as it hides the core files, which they may want to modify.

Either way, I think supporting 1.6.2 and earlier is now not necessary