avikde/koduino #7: SerialUSB functionality

avikde, 2015-11-02:
Need to implement and test this. First candidate will probably be F373 (have some old boards lying around with USB connected). Not sure what VID/PID should be used?

xxxajk, 2015-11-02:
You would need to register for a VID block, which cost something like $5000.
I'm sure if you ask ST nicely, they can donate one to you. I know Atmel does this.
There is also an old business that has a VID block, and will give you a PID (or block of them) free if the project is 100% open source.
Details here:
http://wiki.openmoko.org/wiki/USB_Product_IDs

rogerclarkmelbourne, 2015-11-09:
I've seen some boards that just use USB CDC use STM's default VID PID

BTW. its more complcated than just VID and PID, as to get microsoft to accept your VID's into windows its another big cost

Or you need to do the hack/ workaround that we had to use for the STM32, using locally generated digital certificates.

avikde, 2016-03-28:
SerialUSB writing is now working on the F303 (F3 discovery board used to test) and F373 (tested on Maevarm M4), with support for optional USB disconnect pin. Serial reading has not been tested yet, and changing baud rates does not work (fixed at 115200). TODO: reading tests and support for other boards.

closed 2016-03-28 by avikde