avikde/koduino #11: stm32f103 support

wingspinner, 2015-12-04:
Hi,

Really nice what you have done here. I especially like that you've based this on CMSIS which really reduces the effort to use various STM32 devices. I've be using Roger Clarks STM32Duino for f103 parts with a few of my own modifications which is based on Maple libraries. He's done a fine job and it works quite well but because of the nature of the Maple code it's not easy to modify and doesn't benefit from CMSIS (I assume it's a benefit!). Anyhow, I notice the f103 support in koduino isn't finished and I want to take  a shot at it. Can you give me a quick rundown of what's required to finish up the f103 support to save me some time?

Thx,
Ron Curry

avikde, 2015-12-06:
Hi Ron,
Thanks for getting in touch. The process will go somewhat like this:

1) core must compile with the standard peripheral library for the series.
This can be done by changing the variant to 'f103' in system/Makefile and
running 'make lib' (which builds a compiled object file archive of the std
periph libs). Unfortunately this first step doesn't work yet; the first
reason is that the f1 chip had 'default' and 'remap', just 2 alternate
function modes for gpio configuration, whereas the f3, f4 have 0-15 as
possible alternate functions. Obviously this needs some changes to pinMode,
and (in the worst case) significant code redesign related to gpio
configuration. There may be other areas that require lots of work but
hopefully not.

2) all the boards I use have hardware fpu so I didn't shy away from using
floats in the code for convenience (e.g. see analogWrite). For a chip
without one, perhaps those should be changed to use integer operations
only. I'd maybe skip this to start but you should be aware.

3) the variant.h and variant.cpp in the variants/f103 directory must be
updated to include the correct pin definitions (p.s. if you'd like to use
integer pin names in code just order them so that '0' is whatever pin is
connected to pin 0 on the board of interest, etc), default timer
assignments. I'd generally read through the entire file since it contains a
lot of board specific settings.

4) serial over usb isn't implemented yet :( so if you're using the maple
mini clone might have to do initial debugging through USART and an external
USB-serial device.

5) I didn't make the changes needed to accommodate arduino ide after 1.6.1.
I keep reading that they make changes every release that break how these
third party libraries integrate so I don't think I'll try to keep up with
this for now. Maybe ina few months they will stop changing it and then we
can upgrade our code structure.

That should be enough to get started. Thanks so much for your interest! I
should warn you that because of my low bandwidth for the next months-year
it might take me a long time to merge any major changes (that would require
major testing), but I can definitely help and perhaps we can keep your fork
up to date so it's easy to merge later on.

Thanks,
Avik De

wingspinner, 2015-12-07:
Avik,
Thx! That's a good start. I'm fairly familiar with the guts of the arduino IDE so I was just wanting to understand the current status without digging through all the code. Your update helped. There is existing code to adapt and borrow from - maple/STDuino being one source and perhaps we can get it to a basic working state such that Roger and co may want to merge at some point which would be good for all concerned perhaps. My thinking at the moment is it may be advantageous to leverage CMSIS as you have started. It appears cleaner and should be easier to maintain, add variants, as well as leverage some of the large body of code that already uses CMSIS. Performance I'm not certain sure of but the current maple code has two, three, and perhaps more levels of function calls in some cases for user level functions and perhaps there is a more instruction cycle efficient method of handling interrupts than what maple currently uses. Anyhow, I'll be looking at it over the holidays and see what progress I can make. 

Cheers,

Ron Curry
ron@curry.org

avikde, 2017-05-03:
This may happen (far) in the future; for now I'll focus on F3, F4

closed 2017-05-03 by avikde