avikde/koduino #12: STM32F4 Discovery board on Arduino IDE 1.6.9

cimd, 2016-07-06:
Hi,

Is there a way to make Koduino work on Arduino IDE 1.6.9? It does compile the code, but this is the error I get when trying to upload:
`Warning: platform.txt from core 'Koduino STM32 Boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'Koduino STM32 Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

Sketch uses 23,604 bytes (36%) of program storage space. Maximum is 65,536 bytes.
java.io.IOException: Cannot run program "C:\Program Files (x86)\Arduino/hardware/koduino/stm32/system/stm32ld/stm32ld.exe": CreateProcess error=2, O sistema não pode encontrar o arquivo especificado
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1186)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, O sistema não pode encontrar o arquivo especificado
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
Invalid library found in D:\Arduino\hardware\koduino\stm32\libraries\SDADC_: D:\Arduino\hardware\koduino\stm32\libraries\SDADC_
Invalid library found in D:\Arduino\hardware\koduino\stm32\libraries\SDADC_: D:\Arduino\hardware\koduino\stm32\libraries\SDADC_`

And also, there is no port shown at the Port menu.

avikde, 2016-07-07:
There's probably a way to make it work with 1.6.9...the challenges should be

* It uses a specific version of gcc-arm-none-eabi (4.8.3-2014q1) that comes packaged with 1.6.1 for the Due. If you can put those files in the right spot (or change platform.txt to point to an equivalent toolchain) that should work

* The platform.txt / boards.txt spec may have changed, but hopefully the fixes are small

*  I only tested the F4 core with an F405 and won't claim that things will work on the discovery board. 
The clock settings etc. will probably be wrong unfortunately, since I used a 8 or 16 MHz clock on my custom board.

* The error you got seems to be that it couldn't fine the stm32ld.exe file (this is the serial bootloader I use). If you're using the discovery board I'm assuming it has a built-in ST-LINK. You could try enabling verbose compiler/upload output, at some point it will print the location of the .bin file. You can at that point open up the ST-LINK GUI and upload that .bin file.  If that all works, there's an ST-LINK upload option for the F301 (for example), you could edit boards.txt to enable the same uploader for the F405 variant so you can do this from the Arduino IDE.

Sorry, I'd be more helpful but I'm really short on time to general maintenance of the repo other than adding/maintaining features for the main chips I'm using right now (F301 and F303). :(

cimd, 2016-07-08:
Avik,

Thank you very much for the reply. I'll be travelling for the next few days, and won't be able to test it until next weekend. But I'll do so and advise on the progress on the F4-Disco.

I would appreciate on hearing your thoughts/plans on the development using the SPL, considering that STM32 has discontinued it. Do you ever plan on migrating it to use the new HAL libraries? How much effort would it be required for updating your project to using the HAL? I've seen the work that sheepdoll is doing, on using the HAL libraries. But it seems that the project development is not as active as yours.

avikde, 2016-07-08:
Hi unfortunately I don't think I will do this port to the HAL libraries. I tried once for a few weeks but changes to how the USART driver worked made it almost impossible to reuse the code I had for that module, and I couldn't justify the time to reimplement the data structures (not even just the functions) for the arduino-like libraries.

Besides if you browse through the st forums, there is generally much more experienced support of the SPL, and very little for the HAL libraries (i.e. I'm not sure that many people are using them).

avikde, 2017-05-03:
More recent Arduino IDE support issue is in https://github.com/avikde/koduino/issues/4 closing this one to focus on the other one

closed 2017-05-03 by avikde