avikde/koduino #14: FYI. STM will release a Arduino STM core

rogerclarkmelbourne, 2016-09-12:
Avik

I just thought you may like to know that STM have contacted me and they have developed their own Arduino core for STM32

It is based on the HAL files from the STM32CubeMx to technically it will support F3 (and F4)

My understanding is that they will release a beta later this week, and I've added them as members to the stm32duino github "organization", so the repo will initially be in that account

https://github.com/stm32duino

I have not see the code yet, so I can't give any more details, but I'll keep you posted.

BTW.
Initial release will be F103 L4 F4 and some others possibly L1 (To Be Confirmed)

avikde, 2016-09-12:
Finally! Thanks for the notification. I am a little in trepidation after seeing the general quality of the SPL code (not used HAL as much), but if more people are behind it, it can only be a good thing. I'll keep an ear out for feedback from you.

It will be an enormous project to port all of our code and test it...

rogerclarkmelbourne, 2016-09-12:
Hi Avik

I have not seen the code yet. I only spoke to STM about 12 hours ago (last night for me).

I agree the HAL is a big beast ;-) So it will be interesting to see what they have developed.

BTW
From what I understand STM  commissioned an external company to develop it, and at the moment they have only been testing on Nucleo boards. I don't know how they have been uploading.
I got the impression it was via mass storage, but I can't be sure.

If you want I can add you as a member to stm32duino, but its all going to be public anyway, so you could just fork the repo when its released.

rogerclarkmelbourne, 2016-09-18:
Source for F1 and L4 has been released and is in separate repos in the stm32duino account.

According to Federic at STM, this is the list of other  boards that will be released

- STM32F091RC-Nucleo
- STM32F303RE-Nucleo
- STM32F429ZI-Nucleo
- STM32L053R8-Nucleo

So an F3 version should be released soon.

At the moment the code organisation / structure of the variants seems to be a problem, as their developers didn,t seem to actually consider this.

So we may have to refactor the source a bit (for the variants )

avikde, 2016-09-18:
Thanks Roger, I'll keep an eye on this!

rogerclarkmelbourne, 2016-09-18:
Ok

I'm going to do some minor refactoring of the F103 variants and also add the missing upload tools, later today.

(The only tool to upload they current support is USB Mass storage and not even STLink !)

closed 2017-05-03 by avikde