avikde/koduino #15: Problem with UART.

asipo, 2016-10-12:
Hi.

I am trying to apply Koduino to my custom made board based on STM32F373CCT6.

What I have tested is,

void setup()
{
  Serial1.begin(115200);
  delay(1000);
  
  Serial1 << "Start" << endl;
}

void loop()
{
  Serial1 << "Test" << endl;

  delay(1000);
}

The problems are:
- the baudrate in serial monitor should be set to 57600(half the value set by Serial1.begin(). I tried a couple of other baudrate and results are same(always half the set value)
- the Seiral1 in setup() doesn't work without delay() function, and with delay(<1000). It works with delay(>= 1000). I tried delay time 100, 900, 990 and no use. It seems that the number 1000 is a magic number...

I wonder if you have any idea about these.
Thank you.

avikde, 2016-10-12:
Hi asipo,

* for your first issue, I am fairly certain it's because I compiled the SPL
(into system/libSTM32F37x.a) with the HSE_VALUE defined as 16 MHz, and if
you actually use a different one the rcc files screw up the dividers. I
stopped using the F373 a while back and when I first started, these things
weren't properly done. To fix it, you should try recompiling libSTM32F37x.a
by looking at the comments in system/README.md, and make sure you define
HSE_VALUE to what it actually is.

* for your second issue, I'm not totally sure, but it's probably related to
lines 179-180 in cores/Arduino/USARTClass.cpp. You can try and comment
those out.

Sorry about not being more helpful with this--I'm way too busy to be able
to maintain all the different variants in a way that supports all clock
configurations at the moment :(

asipo, 2016-10-13:
Thank you for your prompt reply.

First of all, the second issue has been solved. It works fine when I commented it out, even though I don't understand why that much amount of time delay is required.

But the first issue, recompiling the library, I was not able to make it.
The makefile in the system folder was modified to

```
# VARIANT = f10x
VARIANT = f37x
# VARIANT = f40_41x
#VARIANT = f301
HSE_VALUE = 8000000UL
UPLOAD_METHOD = SERIAL
UPLOAD_BAUD	= 115200
# LIBRARIES = SPI IMU EEPROM Encoder Brushless Eigen Wire

include $(KODUINO_DIR)/project.mk
```

and set the environment variables to include the path of make and compiler, say 
C:\arduino-1.6.1\hardware\arduino\sam\system\CMSIS\Examples\cmsis_example\gcc_arm for make and
C:\arduino-1.6.1\hardware\tools\gcc-arm-none-eabi-4.8.3-2014q1\bin for compiler

but the when I type "make lib" in the system directory, the error messages are:

```
C:\arduino-1.6.1\hardware\koduino\stm32\system>make lib
----------- something like "wrong syntax"
process_begin: CreateProcess(NULL, basename C:/arduino-1.6.1/hardware/koduino/st
m32/system, ...) failed.
Compiling for f37x, MCU = STM32F37x, project = , upload method is SERIAL
process_begin: CreateProcess(NULL, uname, ...) failed.
process_begin: CreateProcess(NULL, basename C:/arduino-1.6.1/hardware/koduino/st
m32/system, ...) failed.
process_begin: CreateProcess(NULL, basename C:/arduino-1.6.1/hardware/koduino/st
m32/system, ...) failed.
process_begin: CreateProcess(NULL, basename C:/arduino-1.6.1/hardware/koduino/st
m32/system, ...) failed.
"[CC] stm32f37x_adc.c"
Assembler messages:
Fatal error: can't create C:/arduino-1.6.1/hardware/koduino/stm32/system/build/s
tm32f37x_adc.o: No such file or directory
make: *** [C:/arduino-1.6.1/hardware/koduino/stm32/system/build/stm32f37x_adc.o]
 Error 1
```
If you can notice anything wrong, any comments will be appreciated.
Thank you.

avikde, 2016-10-13:
Ok, I just spent some time this morning but since I can only find a working
board with a 16MHz oscillator I may not be able to help you much more.

I added some menu options to the Arduino IDE (see if the 8mhz one does
anything).

You can try using the stm32 clock configuration tool (excel file with
macros) and put the generated file in place of system_sm32f37x.c in
variants/f37x.

I am also able to run 'make lib' fine on my windows computer--you may need
to install make using MSYS, and add the
"arduino-1.6.1\hardware\tools\gcc-arm-none-eabi-4.8.3-2014q1" directory to
your path.

asipo, 2016-10-14:
Hi Avik.
I really appreciate you help.
I was able to "make lib" using mingw make tool and uart baudrate problem was solved.

But the new koduino release doesn't seem to work,

```
Sketch uses 19,932 bytes (7%) of program storage space. Maximum is 262,144 bytes.
python C:\arduino-1.6.1/hardware/koduino/stm32/system/stm32loader.py -p COM13 -b 230400 -y dtr_rts -ew C:\Users\KWJ\AppData\Local\Temp\build7626419891163489296.tmp/sketch_oct14a.cpp.bin 
Traceback (most recent call last):
  File "C:\arduino-1.6.1/hardware/koduino/stm32/system/stm32loader.py", line 31, in <module>
    import serial
ImportError: No module named 'serial'
```
You don't have hardware to test but I just want to let you know.
Anyway I get back to the previous version and very happy now.
Thank you.

luudee, 2016-10-15:
Hi Guys,

I have run into the same issue. I am working to create a 'F405/407 working version.

I was able to fix it for my 'F407 platform. First, as Avikde already pointed out, the SetSysClock
function is all messed up. You have to be very careful editing it ...

But the main reason for the baud rates to be messed up is "RCC_GetClocksFreq". It basically
returns incorrect values for the APBx bus clock.

Once I am done with my fixes, I will send an archive to Avikde, so can integrate it in this library.

But that might take a couple of more days at least ...

Best Regards,
rudi

avikde, 2016-10-15:
I used to edit the system_stm32**.c file by hand but nowadays I never do
it. I always use the STM32 clock configuration xls file.

asipo, 2016-10-17:
Because Koduino doesn't come with hardware, users should be able to generate their own system_stm32**.c file, I believe. For STM32 beginners like me, excel file is much easier to work with.
For STM32F37x series, it seems there is nothing special except APB1 prescaler should equal APB2.

luudee, 2016-10-17:
Avik,

do you have a pointer to that STM32 clock configuration xls file ?

I was actually thinking to write something in tcl/wish, to allow a user to configure his device ...

But I guess a clever xls will do it as well :)

Thanks,
rudi

avikde, 2016-10-17:
http://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-configurators-and-code-generators/stsw-stm32113.html

luudee, 2016-10-17:
Thank you Avik !

closed 2017-05-03 by avikde