avikde/koduino #16: Multiple Source with Makefile

luudee, 2016-10-19:
Hi Avik,

excellent job on these libraries ! Thank you so much for all the hard work !

I have a question regarding multiple sources, when using the makefile approach. 
So I have my sketch, and another part of my code, wich I place in a separate
.c file. It appears as the Makefile tries to compile each .c file as a sketch, e.g.
it does not use -o option on the additional files.

Is there a way around it ? Am I doing something wrong ?

Also, what is the best way to reference 3rd party libraries that I installed in Arduino
tree ?

Lastly, I have made some changes to support 'f407 devices. What's the best way
to submit those ? I would like to send them to you for review and then you can
chose if you want to include them or not. What do you think ?

Many Thanks,
rudi

luudee, 2016-10-19:
Hi Avik,

just a quick follow up. To the first issue I raised:
I found if I rename my file to *.cpp, it is properly processed. But if I leave it as just .c,
it breaks ...

Thanks,
rudi

avikde, 2016-10-19:
Hi rudi,

Thanks for the feedback. Mostly the problem is I'm not a Makefile expert
(even this limited one took a lot of effort), but let me do a few debugging
steps and respond in a few days.

About the F407...are the modifications in the cores/libraries or are they
in separate directories in system or variants? They will be hard for me to
test because I don't have any boards with that chip (and also general lack
of time), but if most of the changes are in variants I'd be happy to merge
them into the repo. What board are you using?

-Avik

luudee, 2016-10-19:
Hi Avik,

ok, no problem, I will think about the makefile and see if I come up with something a bit more flexible, in addition to what you have provided.

About the 'F407, I am using several slightly different boards, and am trying to create a "universal" 'F407 support. My concern is not the 'F407 port - I will check this here myself, my concern is not to break support for the other devices. 

OK, when I am ready to send you something, I will create a diff of all changes and explain what/why they are. And then you can decide if they break anything or not.

Thanks for your help !
rudi

avikde, 2017-05-03:
Abandoning trying to support Makefiles properly I think. I'll move general F407 questions to focus on this issue https://github.com/avikde/koduino/issues/19. SPI and F407 discussion in https://github.com/avikde/koduino/issues/18. I am also going to use SPI with DMA on the F446 in the next few days/weeks so will be testing/committing with that chip.

closed 2017-05-03 by avikde