avikde/koduino #18: SPI Bus and STM32F4

blacktigersoftware, 2017-01-04:
I working on a variant for the STM32F407-Discovery board.  I noticed that in SPI.cpp, line 82, there is a comment for a Steensy default?  Although "Google is my friend", she let me down on a search for Steensy.  Perhaps you are trying to support Teensy 3.x, a Freescale board?  

Either way, I managed to get the SPI working with the onboard accelerometer after a bit of work using the setPins(...) method.  

1) I would like to send you a pull request later this week/next week for this variant.  I've tentatively named it "F407_Discovery".  
2) I would like to change default for SPI1 MISO to PA6
3) Maybe some other changes to the F4 core to more closely match the Discovery board.

I've forked your repo and you are more than welcome to take a look and comment.

blacktigersoftware, 2017-01-10:
FYI:  For those interested - I've got basic blinky working and the built-in accelerometer working with samples on my fork.  Started to do something with the USB.  I like the Teensy 3.x support for USB's and there appears at least to be possible to run a USB Host, USB Client, etc.

xxxajk, 2017-01-10:
If you want to contribute to USB host, please use the new library, UHS 3.0. Don't bother with 2.0 or 1.0, both are a mess.

avikde, 2017-01-10:
Hi blacktiger,

Sorry about the delay. "Steensy" was, a prototype board with a F405 chip with the same nominal footprint and pinout as a Teensy 3 (though it had bugs and I moved straight to application specific boards after that).

Thanks for your work! For the SPI, I'm usually happy to call SPI.setPins, so don't have strong feelings on the default.  Any changes to a new variant are of course fine, but for core ones I'd have to check with the F3 boards. Could you possibly do 2 PR's (one for new variant, and a separate one for core changes)?

Lastly, I've never used USB too much since for communication, Serial -> FTDI gives much higher bandwidth than the VCP implementation I've played with -- see #17 for discussion of its limitations.

Thanks again, sorry for the long delay! I've been so swamped...

palmerr23, 2017-04-19:
Avik,

In making the Black F4VET6 board work, I came across another SPI issue.

The default pin defines for SPI are in C:\Program Files (x86)\Arduino\hardware\koduino\stm32\libraries\SPI\SPI.cpp

**It would be much better for these to be brought up to the variant level to variant.cpp as in most other STM code bases. Similarly for I2C.**

When I tried to make SPI3 work for my board:

```
    else if (SPIx == SPI3) {
		SCK = PB3;
		afSCK = 6;
		MISO = PB4;
		afMISO = 6;
		MOSI = PB5
		afMOSI = 6;
```

it became clear that I was going to break the Teensy pinouts by changing PB4/AF5 to PA6/AF5, which is the SPI1_MISO default for my board
   ```
if (SPIx == SPI1) {
		// Steensy default
                 . . .
		MISO = PA6;
		afMISO = 5;
```

avikde, 2017-04-19:
Hi Richard, I know it isn't optimal where it is now. However, I also find
it hugely important to be able to change this in my "sketch" code so I
defined the SPI.setPins function. Can you just call this in your sketch to
set the pins you want to use?

palmerr23, 2017-04-24:
Avik,

Absolutely.

Clean up can come later.

SPI.setPins will do fine for now!

I'm focusing on ADC/DAC at the moment, so will amend the code as soon as I
have something SPI connected again.

Richard

   _________________________

 0417 514 473
 palmerr23

avikde, 2017-05-03:
I'll be using the F446 with DMA and SPI in the next few days/weeks and will give it a thorough shakedown

avikde, 2017-05-07:
I've been using SPI with DMA (and without, in polling mode) successfully on the F446. I'm not sure about the DMA API's but at least it's all functional. https://github.com/avikde/koduino/commit/01398be91482dab40e69dd4d8d709d503d80dcf2