avikde/koduino #22: LED doesn't blink on STM32F303 Discovery

erwincoumans, 2017-12-18:
Setting the LED on GPIO_Pin_9 to HIGH and LOW should blink one user LED but it doesn't.
I can upload other ELF/BIN programs on this board just fine, using df-util and stlink just fine, and using the Arduino 1.61 IDE shows that the binary is uploaded OK. I also manually uploaded the binary with the same result (no blinking LED).

Could you test it on a STM32F303 Discovery board?
Thanks!
~~~~~~~~~~~~~~~ clang
#include <Arduino.h>
const int led = GPIO_Pin_9;
void setup() {
  pinMode(led, OUTPUT);
  digitalWrite(led, HIGH);
  delay(1000);
  digitalWrite(led, LOW);
  delay(1000);  
}
void loop() {  
}
~~~~~~~~~~~~~~~~~~~~

avikde, 2017-12-19:
This sketch blinks the red LED for me with the master branch. The user LED pin names are listed in the user manual for the F3 discovery, and in koduino literals are defined with the pin names (e.g. in https://github.com/avikde/koduino/blob/master/stm32/variants/f303v/variant.cpp )

```
#include <Arduino.h>
const int led = PE9;
void setup() {
  pinMode(led, OUTPUT);
}
void loop() {
  digitalWrite(led, HIGH);
  delay(1000);
  digitalWrite(led, LOW);
  delay(1000);
}
```

I just selected the ST-LINK upload method and plugged in to the ST-LINK connected to the discovery board.

erwincoumans, 2017-12-19:
Thanks, indeed it works. Someone I assumed GPIO_Pin_9 was the same as PE9 since in https://github.com/nitsky/stm32-example/blob/master/main.c (which worked) seems to use GPIO_Pin_9.
~~~~~~~~~clang
LED3 = 0;
GPIO_PIN[LEDn] = {LED3_PIN, 
#define LED3_PIN                         GPIO_Pin_9
STM_EVAL_LEDInit(LED3);
~~~~~~~~~

Let's close it!

closed 2017-12-19 by erwincoumans