avikde/koduino #24: F303C HSE

arpruss, 2018-05-14:
I am wondering if anybody knows what changes, if any, need to be made to the f303v's hse_8.inc to make it work on the f303c? When I use it without any changes, HSE startup times out. This could be due to a hardware problem on my board (which is a blue pill with the MCU swapped out), but before I solder in a new crystal, I'd like to see if maybe there is something wrong on the software side.

arpruss, 2018-05-15:
Solved: just had to remove the HSEBYP bit from the init code.

closed 2018-05-15 by arpruss

avikde, 2018-05-15:
It's a combination of the CLKSRC_* preprocessor (which can be set through the Arduino menus) and also HSEBYP (which must be changed based on if you have a crystal or an oscillator.

Glad it worked out!